openigtlink/SlicerOpenIGTLink #2: IGT Image messages not accessible as vtkMRMLScalarVolumeNodes

adamrankin, 2018-02-08:
I built Slicer, SlicerOpenIGTLink (and others) and when serving images and transforms from a Plus server images are not available as scalar volume nodes.

What is a vtkMRMLBitStream? Has the way of accessing images changed?

leochan2009, 2018-02-08:
Hi Adam,

vtkMRMLBitStream is a derived class from Vector volume node. 
You could cast it to Scalar volume nodes if needed.
When the SlicerOpenIGTLink module is built with VP9 support(Slicer_USE_VP9 option), the program will use BitstreamNode for both image and video data.  This will bring some benefit if the user want to compress and save the received image(either compressed or uncompressed).
but if SlicerOpenIGTLink is built without VP9 support. Slicer will use Scalar volume for incoming images.

Best,
Longquan

adamrankin, 2018-02-08:
Ok, the default options for SlicerOpenIGTLink enable VP9.

closed 2018-02-08 by adamrankin