openigtlink/SlicerOpenIGTLink #7: IMAGE messages produce multiple nodes

adamrankin, 2018-02-17:
Instead of updating an existing node with new data, a new node is created for each IMAGE message received.

leochan2009, 2018-02-17:
Hi Adam, 
Thanks for reporting the issue,
I try to reproduce the bug, but failed.
I use ImageServer program in the openigtlink as server.
and directly connect to the server from Slicer. Only one image node was created and was updated correctly with each image message.
Also i tried also deactivating the Videostreaming in SlicerOpenIGTlink library, the module also works well.
Would you show me the details to reproduce the error?

Best,
Longquan

adamrankin, 2018-02-17:
Slicer 6024dfebc9555026a937f4a3fa67f1013e49b753
cmake -G "Visual Studio 14 2015 Win64" -DQt5_DIR=D:/Qt/5.9.1/msvc2015_64/lib/cmake/Qt5 ../S4 -DSlicer_USE_SimpleITK:BOOL=OFF

SlicerOpenIGT f7055efc9fd995f7222c62bced6f2b2a77b3af12
USE_VP9:BOOL=OFF

PlusLib 83764808e0c405b06f2ba4b5f5d451703afcdd35
Config file PlusDeviceSet_Server_MmfColorVideoCapture.xml

adamrankin, 2018-02-17:
All built in release

adamrankin, 2018-02-17:
Ah, issue only occurs with VectorVolume. The issue is colour image streaming.

leochan2009, 2018-02-17:
Hi Adam, 

i built the Module in Release and test again using ImageServer that produces images message with number of component equal 3 
(just add one line the ImageServer.cxx file, imgMsg->SetNumComponents(3) )
Slicer : 5d294412119bf51f14c3a19ec98f055c1cbe30fe
SlicerOpenIGTLink: master
Use_VP9 = OFF
.In this way the slicer will use VectorVolume for incoming image message.
In my test, Slicer generated VectorVolume correctly and updating the node was also correct.
Probably i need to have a test on the Windows machines under the plus environment again when i am back in Office on Tuesday. See the attach photo for more information.
[image]

BTW, i actually found another bug related with node modified event. Just commit the change to github. 

Best,
Longquan

leochan2009, 2018-02-20:
@adamrankin 
HI,
I reproduced the error your mentioned. The bug was fixed in the later commits.
If you update the SlicerOpenIGTLink to the latest commit, the issue will be solved.
Thanks for reporting the Bug!

Best,
Longquan

closed 2018-02-20 by leochan2009