openigtlink/SlicerOpenIGTLink #8: Storing all image devices in vtkMRMLBitStreamNodes

Sunderlandkyl, 2018-02-27:
Sorry if the answer is obvious, but I don't understand why all image devices use vtkMRMLBitStreamNode.

https://github.com/openigtlink/SlicerOpenIGTLink/blob/0d343f128312e4e3e28a754fe186f8675eb0b7ee/OpenIGTLinkIF/MRML/vtkMRMLIGTLConnectorNode.cxx#L356-L366

Is it so that they can be saved as compressed video using Sequences?

leochan2009, 2018-02-27:
Hi Kyle,
thanks for raising the question.
Actually Adam has the same concern, as change has effect for other modules that assume the incoming image will be represented as "vtkMRMLVolumeNode"  in slicer.  
Please refer to this issue for more details: https://github.com/openigtlink/SlicerOpenIGTLink/issues/2
Yes, as Tamas mentioned during the Project week, using bitstreamnode might be very helpful for video data storage. But now many modules will be affected by this change, I need a in-depth discussion...

Sunderlandkyl, 2018-02-27:
Ah, I didn't see the previous issue.

Specifically, I just noticed that the IJKToRAS was not set for vtkMRMLBitStreamNodes created from Image devices, which was what led me to realizing that they were not volume nodes.

I made a pull request setting IJKToRAS here: #9

I haven't encountered any other application specific issues yet.

leochan2009, 2018-02-28:
Thanks for finding the bug :-)

closed 2018-02-28 by leochan2009