openigtlink/SlicerOpenIGTLink #18: Cannot load qSlicerOpenIGTLinkIFModule.dll

tavaughan, 2018-03-15:
I built SlicerOpenIGTLink on Windows from source. When I try to load it in Slicer I get the following error message:

>   Error(s):
>    Cannot load library C:/D/SOIGTLD/inner-build/lib/Slicer-4.9/qt-loadable-modules/Debug/qSlicerOpenIGTLinkIFModule.dll: The specified module could not be found.

I set the module path to:
inner-build/lib/Slicer-4.9/qt-loadable-modules/Debug/

I built:
- 3D Slicer, master branch commit 3c16aacbdf2fb158527eab17ab7e1cab7b9a4914
- 3D Slicer was built with Qt 4.8.7
- Latest (at the time of writing) SlicerOpenIGTLink commit 9c32267c7dfad43ea5be262f5cf96210224b1152

I consulted with @Sunderlandkyl 
We tried copying the files from inner-build/bin/Debug/ to inner-build/lib/Slicer-4.9/qt-loadable-modules/Debug/
That seemed to solve my issue for now. Is this a bug?

leochan2009, 2018-03-15:
i have successfully built SlicerOpenIGTLink in windows platform before, the reason why it failed now might due to the slicer updates, which doesn't set the output directory correctly.  I will take a look soon. Hopefully come back to you tomorrow.

tavaughan, 2018-03-15:
Ok, thanks.

I do not have this issue in Linux, but I'm also using an older Slicer (March 1).

lassoan, 2018-03-16:
When an extension uses third-party libraries (SlicerOpenIGTLink does) then you need to start Slicer with specifying additional launcher settings:

    Slicer.exe --launcher-additional-settings c:\D\SlicerOpenIGTLink-win64rel\inner-build\AdditionalLauncherSettings.ini

tavaughan, 2018-03-16:
Thanks @lassoan , yes it looks like that fixed my issue.

I assume then that for development builds in Windows, this will always be necessary.

leochan2009, 2018-03-16:
Great that the issue is fixed.
I just had a local build of Slicer, and SlicerOpenIGTLink,
The build looks fine. qSlicerOpenIGTLinkIFModule.dll is located at the correct place.
However, I got a similar error message:

  Error(s):
    Cannot load library C:\Users\leochan\Desktop\Work\SlicerOpenIGTLink-build\inner-build\lib\Slicer-4.9\qt-loadable-modules\Debug\qSlicerOpenIGTLinkIFModule.dll: The specified module could not be found.
```
Traceback (most recent call last):
  File "C:/Users/leochan/Desktop/Work/SlicerB/Slicer-build/lib/Slicer-4.9/qt-scripted-modules/DataProbe.py", line 30, in __init__
    if slicer.mrmlScene.GetTagByClassName( "vtkMRMLScriptedModuleNode" ) != 'ScriptedModule':
AttributeError: 'vtkCommonCorePython.vtkObject' object has no attribute 'GetTagByClassName'
```