openigtlink/SlicerOpenIGTLink #20: Automatic test fails on Windows

lassoan, 2018-03-16:
Automatic test fails on Windows, which on Windows prevents building of all dependent extensions (such as SlicerIGT).

The test does not fail in debug mode, but crashes in Release mode. To debug it, we would need to build in RelWithDebInfo mode, hopefully the problem is reproducible that way. If not, then we may add log messages and see what crashes exactly.

If we cannot figure out what the issue is, then in the short term we should disable the test on Windows.

lassoan, 2018-03-16:
@Sunderlandkyl Can you have a look at this? It would be important to make SlicerIGT available in the nightly builds.

Sunderlandkyl, 2018-03-16:
Well, when I tried to run the tests on windows yesterday, the test executables were in the bin directory, while the dlls were all in the lib directory, causing the test to fail.

Is there a CDash report for the tests?

Sunderlandkyl, 2018-03-16:
@lassoan The test runs fine in release mode for me, unless there is another server already running on the same computer in which case it will connect to the wrong server and crash.

I'm going to debug that crash now, but if that is the issue, then the test will likely still fail.

Sunderlandkyl, 2018-03-16:
@lassoan @leochan2009 #21 fixes the issue that I had with the test crashing when I had a Plus server running on the same computer, but I haven't been able to replicate any crashes otherwise.

lassoan, 2018-03-16:
Thanks a lot @Sunderlandkyl it fixed the issue!! It's interesting, because I don't think I had other PlusServer running on my computer (or on the factory computer). How did you figure out that this was the problem?

Sunderlandkyl, 2018-03-16:
I happened to have a plus server running when I went to run the test, causing it to crash. Just luck I guess.

Sunderlandkyl, 2018-03-16:
@lassoan On a related note, would it make sense to change the port for the test from 18944 to a different, less commonly used port?

lassoan, 2018-03-16:
Yes, probably it would be better to avoid potential conflicts with other PlusServer instances.

leochan2009, 2018-03-16:
Interesting findings! I will merge your commits.
Hopefully the test on the server will be fixed. finger crossed!

closed 2018-03-18 by leochan2009

lassoan, 2018-03-18:
This morning there are no more test errors on SlicerOpenIGTLink dashboard and SlicerIGT extension is successfully built on Windows!

PR #23 (2018-03-18): Fix test missing additional launcher settings
This should fix #20