openigtlink/SlicerOpenIGTLink #24: Moving OpenIGTLinkRemote and PlusRemote into this extension

lassoan, 2018-03-19:
Since OpenIGTLinkIF is in a separate extension now, we should move OpenIGTLinkRemote and PlusRemote extensions here. They belong to each other - a change in SlicerOpenIGTLinkIF is likely to impact OpenIGTLinkRemote and PlusRemote as well, and this would remove the hard dependency of SlicerIGT extension on SlicerOpenIGTLink extension. See also https://github.com/SlicerIGT/SlicerIGT/issues/173.

@leochan2009 Do you agree? If yes, then we'll send a pull request that contains these two modules.

leochan2009, 2018-03-19:
It make sense to move the OpenIGTLinkRemote to SlicerOpenIGTLink.
Regarding the PlusRemote module, i think we need to rename it to make it not Plus-related.
Please send the pull request, i will review it,
Thanks,
Longquan

Sunderlandkyl, 2018-03-19:
I added the migration of PlusRemote and OpenIGTLinkRemote to #19.

Any suggestions on what to rename PlusRemote?

lassoan, 2018-03-19:
Unlike OpenIGTLinkRemote, PlusRemote is not a general remote control module, but it is developed specifically to work with Plus. I don't think it should be renamed.

leochan2009, 2018-03-19:
En, this is just a personal opinion. Previously I had the inclination of keeping the library to itself, and only include the modules that are generalized,  and move the PlusRemote to another extension.
After talking with Junichi,  we are fine with the expanding of SlicerOpenIGTLink extension.
probably in the future, we could even include MatlabBridge into SlicerOpenIGTLink
I will review the change and merge the commit soon

closed 2018-03-21 by leochan2009