openigtlink/SlicerOpenIGTLink #25: Slicer crashes when streaming webcam to Slicer from Plus

Sunderlandkyl, 2018-04-17:
I've found that streaming from a webcam to Slicer using Plus will cause Slicer to crash when using the latest nightly (4.9.0-2018-04-16). It doesn't seem to happen with other types of images. Plus continues to run without error.

It seems to happen reliably, although the crash doesn't always happen right away. I've replicated the crash using both the color and greyscale MMF config files (https://github.com/PlusToolkit/PlusLibData/blob/master/ConfigFiles/PlusDeviceSet_Server_MmfVideoCapture.xml and https://github.com/PlusToolkit/PlusLibData/blob/master/ConfigFiles/PlusDeviceSet_Server_MmfColorVideoCapture.xml)

When I debugged SlicerOpenIGTLink using Visual Studio, it crashes here:
https://github.com/openigtlink/OpenIGTLink/blob/54ea5a0ce55b8280f363e24f74eb3c1bd40ed5a1/Source/igtlMessageBase.cxx#L760

There are no remarkable messages in the log that I can see: ([Slicer_27152_20180417_153715.log](https://github.com/openigtlink/SlicerOpenIGTLink/files/1921314/Slicer_27152_20180417_153715.log))

I'll look into this some more, but I'm not really that familiar with base OpenIGTLink.
Any suggestions as to possible causes would be appreciated.

Sunderlandkyl, 2018-04-17:
I should clarify that this is on Windows. I'll try to test on Linux and MacOS.

leochan2009, 2018-04-18:
Hi Kyle,
Thanks for reporting! i will take a look ASAP

leochan2009, 2018-04-18:
Tried on the nightly build on April 1st, no crash happens, now will check on the latest nightly build

Sunderlandkyl, 2018-04-18:
It's quite strange. I can still replicate the issue on Windows, but not on Mac anymore.

leochan2009, 2018-04-19:
I reproduced the same bug in my Windows machine too.  Currently working on it

leochan2009, 2018-05-01:
Bug found, let's check the nightly build tomorrow!

Sunderlandkyl, 2018-05-01:
Thanks for taking a look at this Longquan!

leochan2009, 2018-05-02:
Fixed

closed 2018-05-02 by leochan2009