openigtlink/SlicerOpenIGTLink #31: Qt verson

leochan2009, 2018-05-17:
Hi, Kyle,
I just found some QT version issue. the remote modules won't compile with older version of Slicer, as the older Slicer are using Qt 4.8.7.
I assume some researchers might try to compile SlicerOpenIGTLink with older Slicer, so we should achieve some backward compatibility.  do you think so?
@Sunderlandkyl

Sunderlandkyl, 2018-05-17:
Possibly, although the older versions of Slicer already come bundled with OpenIGTLinkIF. I guess that the old extension doesn't contain support for compressed video, so I can see why users might want to upgrade to the new extension.

If you think that it would be a good idea to support Qt 4, I can go through the remote modules and try to restore compatibility.

leochan2009, 2018-05-17:
Hi, Kyle,
i already did the compatibility, just a few lines :-)

leochan2009, 2018-05-17:
but i need your review, because it is your module, i hope the modification won't break anything:-)

Sunderlandkyl, 2018-05-17:
I think It should be fine. I don't imagine that it will break anything.

closed 2018-05-23 by leochan2009