openigtlink/SlicerOpenIGTLink #55: Filename without file extension in PlusRemote recording crashes PlusServer

jamesobutler, 2018-08-19:
**Version Details:**
Windows 7
Slicer 4.9.0-2018-08-17 and installed SlicerOpenIGTLink extension
Plus-2.7.0.ecfe7933 - Win64

**Issue:**
If you specify a filename without a file extension in the PlusRemote recording section, it will crash PlusServer:
`ERROR|011.172000| No writer for file: Recording| in E:\D\PTNP64b\PlusLib\src\PlusCommon\IO\vtkPlusSequenceIO.cxx(99)`
[image]

In Plus, if you stop recording and specify a output-file without an extension it will give a warning and choose the nrrd extension type.
`WARNING|025.160000| Writing sequence file with '' extension is not supported. Using nrrd extension instead.| in E:\D\PTNP64b\PlusLib\src\PlusCommon\IO\vtkPlusNrrdSequenceIO.cxx(1300)`

Sunderlandkyl, 2018-08-19:
Interesting. Thanks, I'll take a look at it.

jamesobutler, 2018-08-19:
Thanks!

Also, Filename, Add timestamp to filename, and Compression widgets should probably be disabled if actively recording, since changing the properties of those widgets such as updating the filename aren't actually used when stop recording is pressed.  They are only used the next time Start Recording is pressed.

Sunderlandkyl, 2018-08-20:
The crash should be fixed by: https://github.com/PlusToolkit/PlusLib/commit/78c653212798df9d5452d488e218443258daa163

jamesobutler, 2018-08-20:
Just tested and that solved the PlusServer crash. :+1: Thanks!

There's really no indication when using the PlusRemote module that a filename recording issue occurred other than looking at the Plus log messages in another window.  Log below
```log
time|level|timeoffset|message|location
082018_182235.091|INFO|000.002000| System start timestamp: 894| in c:\p\pr17_x64\pluslib\src\pluscommon\vtkplusaccuratetimer.cxx(51)
082018_182235.473|INFO|000.384000| Software version: Plus-2.7.0.78c65321 - Win64| in vtkPlusLogger(188)
082018_182235.473|INFO|000.385000| Logging at level 3 (INFO) to file: C:/Users/jobutler/PlusApp-2.7.0.20180820-Win64/data/082018_182235_PlusLog.txt| in c:\p\pr17_x64\pluslib\src\plusserver\tools\plusserver.cxx(81)
082018_182235.934|INFO|000.845000| Selected US image orientation: MF| in c:\p\pr17_x64\pluslib\src\plusdatacollection\vtkplusdatasource.cxx(301)
082018_182236.188|INFO|001.100000| Server status: Reading configuration.| in c:\p\pr17_x64\pluslib\src\plusserver\tools\plusserver.cxx(92)
082018_182236.240|INFO|001.152000| Server status: Connecting to devices.| in c:\p\pr17_x64\pluslib\src\plusserver\tools\plusserver.cxx(103)
082018_182238.558|INFO|003.469000| Server status: Starting servers.| in c:\p\pr17_x64\pluslib\src\plusserver\tools\plusserver.cxx(116)
082018_182238.970|INFO|003.881000| Server status: Server(s) are running.| in c:\p\pr17_x64\pluslib\src\plusserver\tools\plusserver.cxx(147)
082018_182238.971|INFO|003.883000| Press Ctrl-C to quit.| in c:\p\pr17_x64\pluslib\src\plusserver\tools\plusserver.cxx(148)
082018_182238.993|INFO|003.904000| Plus OpenIGTLink server listening on IPs: 192.168.0.103, 169.254.112.2, 127.0.0.1 -- port 18944| in c:\p\pr17_x64\pluslib\src\plusserver\vtkplusopenigtlinkserverwin32.cxx(81)
082018_182608.549|INFO|213.461000| Received new client connection (client 1 at 127.0.0.1:18944). Number of connected clients: 1| in c:\p\pr17_x64\pluslib\src\plusserver\vtkplusopenigtlinkserver.cxx(257)
082018_182608.628|INFO|213.540000| OpenIGTLink broadcasting started. No data was available between 0-204.826sec, therefore no data were broadcasted during this time period.| in c:\p\pr17_x64\pluslib\src\plusserver\vtkplusopenigtlinkserver.cxx(408)
082018_182617.277|INFO|222.189000| vtkPlusStartStopRecordingCommand::Execute:| in c:\p\pr17_x64\pluslib\src\plusserver\commands\vtkplusstartstoprecordingcommand.cxx(263)
082018_182617.277|INFO|222.189000| vtkPlusStartStopRecordingCommand::Execute: StartRecording| in c:\p\pr17_x64\pluslib\src\plusserver\commands\vtkplusstartstoprecordingcommand.cxx(289)
082018_182617.279|ERROR|222.190000| No writer for file: Recording| in c:\p\pr17_x64\pluslib\src\pluscommon\io\vtkplussequenceio.cxx(100)
082018_182617.279|ERROR|222.190000| Could not create writer for file: Recording| in c:\p\pr17_x64\pluslib\src\plusdatacollection\virtualdevices\vtkplusvirtualcapture.cxx(206)
082018_182617.279|ERROR|222.190000| Command execution failed| in c:\p\pr17_x64\pluslib\src\plusserver\vtkpluscommandprocessor.cxx(170)
082018_182622.250|INFO|227.161000| vtkPlusStartStopRecordingCommand::Execute:| in c:\p\pr17_x64\pluslib\src\plusserver\commands\vtkplusstartstoprecordingcommand.cxx(263)
082018_182622.250|INFO|227.161000| vtkPlusStartStopRecordingCommand::Execute: StopRecording| in c:\p\pr17_x64\pluslib\src\plusserver\commands\vtkplusstartstoprecordingcommand.cxx(289)
082018_182622.250|ERROR|227.162000| Command execution failed| in c:\p\pr17_x64\pluslib\src\plusserver\vtkpluscommandprocessor.cxx(170)
```