openigtlink/SlicerOpenIGTLink #57: Non-responsive Slicer on close if there is an active OpenIGTLinkIF connector

jamesobutler, 2018-08-20:
Windows 7
Slicer 4.9.0-2018-08-17

If I open Slicer and go to OpenIGTLinkIF module, add an IGTLConnector and set the status to Active, then close Slicer, Slicer will become unresponsive on close displaying a vtkOutputWindow with the following messages:

````
Generic Warning: In D:\D\P\Slicer-0-buuild\VTKv9\Common\Core\vtkDebugLeaks.cxx, line 281
Deleting unknown object: vtkMRMLNodeSequencer
````

jamesobutler, 2018-08-21:
Apologies, after coming back to this it appears that vtkMRMLNodeSequencer is introduced with the [Sequences](https://github.com/SlicerRt/Sequences) extension. After uninstalling the Sequences extension, following the steps listed in my original post resulted in no Slicer issues when closing.

However, this issue has so far only presented itself when doing the steps stated in the original message with the Sequences extension also being installed.  Maybe there is an issue with how the SlicerOpenIGTLink extension is handling removing nodes on close? Or maybe an issue with how the Sequences extension is handling removing other nodes on close?  I have been unable to recreate this issue when strictly using the Sequences extension.

I don't have a Slicer-build right now on my Windows 7 machine to debug this further beyond what output I see using the Slicer-nightly binary.

Sunderlandkyl, 2018-09-04:
I actually have recently experienced this problem.

It's interesting because it only happens on some computers. Sequences doesn't seem to be a factor as far as I can tell.

I couldn't get a debug build to cause the issue on my computer, but I tested a nightly build on a different computer that has the issue and it looks like it's an issue with the igtlioConnector not terminating correctly.

I'll take a closer look at it.

commit d1bf5aba (2018-11-01): fix #57: Call Stop on all connector nodes when the application is closing

- Currently stop is called on the OpenIGTLinkIO connectors when the connector nodes are deconstructed
- This happens too late in the deconstruction process, breaks the thread handling mechanism
- Stop is now called during the vtkSlicerOpenIGTLinkIFLogic deconstructor
commit 3324f672 (2018-11-01): fix #57: Call Stop on all connector nodes when the application is closing

- Currently stop is called on the OpenIGTLinkIO connectors when the connector nodes are deconstructed
- This happens too late in the deconstruction process, breaks the thread handling mechanism
- Stop is now called during the vtkSlicerOpenIGTLinkIFLogic deconstructor
closed 2018-11-01 by Sunderlandkyl