openigtlink/SlicerOpenIGTLink #58: Missing modules in Linux, Slicer nightly build

tavaughan, 2018-08-22:
Steps I did in Ubuntu 18.04:
1. Download latest nightly
2. Unzip Slicer
3. Go to extension manager
4. Install OpenIGTLink
5. Restart Slicer
6. Look for PLUSRemote, OpenIGTLinkIF, or other modules

leochan2009, 2018-09-20:
The .so files (OpenIGTLink and OpenIGTLinkIO) are located in the wrong location. copy the files to the location as shown in the screenshot will solve the problem. Will dig into the problem.
[image]

MaXZon3, 2019-01-07:
> The .so files (OpenIGTLink and OpenIGTLinkIO) are located in the wrong location. copy the files to the location as shown in the screenshot will solve the problem. Will dig into the problem.

I had the same issue and this quick fix works. :sweat_smile:

Sunderlandkyl, 2019-01-11:
Fixed by 9e0f3f3

closed 2019-01-11 by Sunderlandkyl