openigtlink/SlicerOpenIGTLink #61: Incoming Images flicker when using SlicerIGT Volume Reslice Driver

Sunderlandkyl, 2018-09-04:
[image]

Although ctkVTKAbstractView::scheduleRender is not called during vtkMRMLIGTLConnectorNode::PeriodicProcess when using the changes from https://github.com/openigtlink/SlicerOpenIGTLink/pull/60 and https://github.com/Slicer/Slicer/pull/1011, the flickering still appears to be a problem.

lassoan, 2018-09-04:
Let's add logs to vtkMRMLSliceLayerLogic.cxx to see that when the transforms or image changes then reslicing transform (linearXYToIJKTransform) does not contain any significant out-of-plane offset or rotation in (after SnapToPermuteMatrix call, linearXYToIJKTransform should only contain axis reordering and scaling). It would confirm that flickering is due to misaligned slice view and volume.

Sunderlandkyl, 2018-09-04:
Things to note:

ImageModifiedEvent is still called at the igtlio level when processing incoming messages, so it is not caught by incomingNode->StartModify() in vtkMRMLIGTLConnectorNode::vtkInternal::ProcessIncomingDeviceModifiedEvent

But...
The flickering is still present even when calling StartModify() on all nodes before calling IOConnector->PeriodicProcess().

There are only 2 places where scheduleRender is triggered (that I've been able to find):
1. As a result of calling EndModify on each of the incoming nodes.
2. At some Qt level (see below)

[image]

Still debugging... let me know if you have any other theories. @lassoan @tavaughan

lassoan, 2018-09-04:
> At some Qt level (see below)

This is most probably result of a single-shot timer calling scheduleRender. Search for "scheduleRender" in all files in source code of Slicer and all installed extensions (all C++ and py files) to find out which may make the call. There should not be too many occurrences.

Maybe it's the slice view annotations ("B: Image_Chest")? That component is implemented very poorly, so I would not be surprised if it had side effects.

Sunderlandkyl, 2018-09-04:
> after SnapToPermuteMatrix call, linearXYToIJKTransform should only contain axis reordering and scaling

Experimenting with this I've found that the linearXYToIJKTransform is not always rotation free, Ex.:
```
 1.4663643266049655 0.0019934533803873666 0.0024338962422760235 -279.61204849653495
-0.0021076203383660341 1.5470749512238560 0.0012449746477579682 -123.42141582088425
-0.0035670355681541865 -0.0017353879449513721 0.99999642735797878 -10.180861958594278
 0.00000000000000000 0.00000000000000000 0.00000000000000000 1.0000000000000000
```

It Seems to be that when EndModify() is called, there are two separate UpdateTransforms() calls.
1. From volume node ModifiedEvent -- This can sometimes cause matrices with rotation like the one shown above
1. From volume reslice driver -- Seems to always be rotation free

Still investigating.

Sunderlandkyl, 2018-09-04:
Here's my current hypothesis:
- Calling Node->EndModify() triggers an update in vtkMRMLSliceLayerLogic, which causes a render event at the old position. Since the image is no longer at that location, it flickers to black.
- The VolumeResliceDriver is then triggered, which causes a second update in vtkMRMLSliceLayerLogic, this time at the correct position.

Possible fix:
Set the priority on the VolumeResliceDriver observers to some value >0.0 so that the volume reslice driver is updated first. There will still be two render events, but at least they'll be at the same location.

tavaughan, 2018-09-05:
>> At some Qt level (see below)

Search found three occurrences of "scheduleRender" in 3D Slicer source code (excluding those related to segmentations, data probe, and annotations). I did not see any in extensions.
https://github.com/Slicer/Slicer/blob/c88ee87ef2c1864fc5d737ef3654b64c2998dd91/Libs/MRML/Widgets/qMRMLThreeDView.cxx#L133
https://github.com/Slicer/Slicer/blob/c88ee87ef2c1864fc5d737ef3654b64c2998dd91/Libs/MRML/Widgets/qMRMLSliceView.cxx#L196
https://github.com/Slicer/Slicer/blob/c88ee87ef2c1864fc5d737ef3654b64c2998dd91/Libs/MRML/Widgets/qMRMLSliceWidget.cxx#L75

Sunderlandkyl, 2018-09-05:
Well the (potentially) good news is that I wasn't able to reproduce the issue after implementing the changes (see 
https://github.com/Sunderlandkyl/SlicerIGT/commit/65304b0d7d2364524863184284748efa6bd3166c).
>Set the priority on the VolumeResliceDriver observers to some value >0.0 so that the volume reslice driver is updated first. There will still be two render events, but at least they'll be at the same location.

This also requires the addition of the vtkSetAndObserveMRMLNodeEventsWithPrioritiesMacro macro in Slicer.

Does this seem like a reasonable solution?

Sunderlandkyl, 2018-09-05:
Macro is here:
https://github.com/Sunderlandkyl/Slicer/commit/02a792a1ee524aba4084adb141eb5c0470f51023

lassoan, 2018-09-05:
Tuning priorities may be a good temporary workaround, but it is not suitable for a long-term solution, as it just masks a problem and similar issues may come up in the future when new functionality is added or used in a new combination.

Sunderlandkyl, 2018-09-05:
After discussion with Andras it was decided that there should be a pauseRender() function in ctkVTKAbstractView that can temporarily pause rendering (prevents scheduleRender from triggering forceRender).

Sunderlandkyl, 2018-09-05:
On the topic of handling pauseRender() in ctkLayoutManager.
- Not all views will share the same pause state, in which case, the ctkLayoutManager::pauseRender() function cannot simply return a bool, but rather some sort of list of the previous states for all ctkVTKAbstractView in the current layout

But that leads to:
- What if the layout is changed while rendering is paused, so that the returned list is no longer valid

lassoan, 2018-09-05:
Good point. We can then add pauseRender() resumeRender() methods and save detailed state in the layout manager. It would be similar to state management in Qt's setOverrideCursor/restoreOverrideCursor (https://doc.qt.io/archives/4.6/qapplication.html#setOverrideCursor).

If layout is changed then the layout manager could restore render state of views that are removed and set state of views that are newly added.

Sunderlandkyl, 2018-10-16:
This issue has been resolved.

closed 2018-10-16 by Sunderlandkyl