openigtlink/SlicerOpenIGTLink #67: [Feature Request] Sending fiducial points/Markup points via OpenIGTLinkIF

chaicl95, 2019-02-02:
I would like to request a new feature where the module can send and receive fiducial points/markup points via OpenIGTLinkIF. I found that the OpenIGTLink Library has the capability to send point with 3D coordinates as PointMessage. 
For my knowledge, this feature is not implemented currently. It would be nice to have this feature since my current project required me to send a group of 3D coordinate points into Slicer as markup points to perform landmark registration.

leochan2009, 2019-02-02:
Hi,

You can use transformation message to send 3D coordinates points. In slicer there will be vtkmrmltransformationnode created. You can then apply the node to fiducial markups

chaicl95, 2019-02-02:
I see. Is there maximum number of transformation message that I can send in one go because I need to send quite a number of points (around 80)?

leochan2009, 2019-02-02:
you only send one message at one time. But you could send them in a for loop. You will need to align each transformation node with each fiducial in slicer. Each transformation node in slicer is created automatically when a transformation message is received. The node will be updated automatically with upcoming message. When you send the message, make sure you have the name setup from the sender side. The name is used for node creation and update.

chaicl95, 2019-02-02:
Thanks for the info. I'll try it.

leochan2009, 2019-02-02:
https://github.com/SlicerIGT/SlicerIGT/tree/master/FiducialRegistrationWizard

Here is some examples for fiducial registration, hope it helps!

adamrankin, 2019-02-02:
You could also look into [TDATA](https://github.com/openigtlink/OpenIGTLink/blob/master/Documents/Protocol/trackingdata.md) messages, as they can send arbitrary number of transforms in a single message.

lassoan, 2019-02-02:
The described workaround is acceptable for some use cases, but at some point we'll have to implement transfer of fiducial list as POINTS message.

Shouldn't we keep this ticket open as a feature request?

tokjun, 2019-03-20:
I agree with @lassoan . I've just realized that [ROS-IGTL-Bridge](https://github.com/openigtlink/ROS-IGTL-Bridge) is not fully working with the new OpenIGTLinkIF because of this limitation.

ChiaraTe, 2019-05-20:
HI, why is this option possible in slicer 4.8.1 and not in slicer 4.10.1?

angad15, 2020-01-19:
Same problem as @tokjun. ROS-IGTL-Bridge is not fully working.

adamrankin, 2020-08-25:
After acquiring an Intel RealSense camera, I realize this is needed to send point clouds to Slicer. I'll start working on it.

lassoan, 2020-08-25:
We send the RealSense point cloud as depth image and convert to polydata in Slicer using this module: https://github.com/PerkLab/DepthImageToPointCloud. Sending the point cloud in a polydata message would be more general, but it is more efficient this way.

This ticket is about fiducials, which is typically just a few ten points, may be up to a few thousand, and they are converted to markup fiducials in Slicer.

closed 2020-11-20 by lassoan

lassoan, 2020-11-21:
This is implemented now.

commit 576d3dde (2020-11-20): ENH: Add support for transferring markups fiducial node as POINT message

Fixes #67