openigtlink/SlicerOpenIGTLink #72: Missing methods from OpenIGTLinkRemoteLogic in Slicer 4.10

ageorgou, 2019-03-05:
I am not sure whether this is an issue with the source or with the extension build process. I cannot access the `SendCommand` and `CancelCommand` methods of `vtkSlicerOpenIGTLinkRemoteLogic` on Slicer 4.10.

Steps to reproduce:
- Install Slicer 4.10
- Install the OpenIGTLink extension
- Try the following in the Slicer console:
```python
>>> slicer.modules.openigtlinkremote.logic().SendCommand
Traceback (most recent call last):
  File "<console>", line 1, in <module>
AttributeError: 'vtkSlicerOpenIGTLinkRemoteModuleLogicPython.vtkSli' object has no attribute 'SendCommand'
```

Indeed, checking `dir(slicer.modules.openigtlinkremote.logic())` does not show either `SendCommand` or `CancelCommand`. In contrast, trying the above on Slicer 4.8 does include those methods, and I have managed to send commands successfully.

It also seems that the two missing methods are the only public ones which take a `igtlioCommand` - maybe the change from  `vtkSlicerOpenIGTLinkCommand` has something to do with it?

Thanks in advance for any help!

Sunderlandkyl, 2019-03-05:
Yes, unfortunately functions containing igtlioCommand aren't wrapped in Python.

I can see about restoring the previous interface.
What signature did you normally use?
1. SendCommand(std::string device_id, std::string command, std::string content, bool blocking = true, double timeout_s = 5)
2. SendCommand(vtkSlicerOpenIGTLinkCommand* command)

ageorgou, 2019-03-06:
Thanks for clarifying. We're using the method on the logic, not the connector node directly, with the signature:
`SendCommand(vtkSlicerOpenIGTLinkCommand* command, const char* connectorNodeId)`

Sunderlandkyl, 2019-03-06:
Missing function signatures have been restored in b3c1005.

ageorgou, 2019-03-06:
Thanks @Sunderlandkyl! Will this make its way into the extension on the Slicer catalogue? (sorry, I'm not sure how often/automatically the extensions are update)

Sunderlandkyl, 2019-03-06:
The fix will be included in the extension manager of both the stable (4.10.1) and nightly versions tomorrow morning.

ageorgou, 2019-03-11:
Can confirm the methods are restored in 4.10.1. Thanks again!

closed 2019-03-11 by ageorgou