openigtlink/SlicerOpenIGTLink #73: Plus Remote: Slicer freezes when receiving reconstructed volume on Linux

Sunderlandkyl, 2019-03-11:
Original report here:
https://discourse.slicer.org/t/slicer-4-10-1-for-linux-crashes-when-using-live-reconstruction-in-module-plus-remote/6105

closed 2019-03-11 by Sunderlandkyl

lassoan, 2019-03-11:
I'm just curious, why did it hang and did you fix it? (it was not obvious from the source code changes)

Sunderlandkyl, 2019-03-11:
The mutex at the top of the function was the problem.

Plus Remote received the CommandCompletedEvent, and then tried to send a follow-up command (Volume reconstruction).

I just separated the function into two blocks so that the command is completed and removed from the deque, the mutex is unlock, and then the CommandCompletedEvent is invoked. This allows observers to trigger follow-up commands without issue/hassle.

I just realized that the commit message is not correct. It wasn't the response, it was the ability to send follow-up commands, not responses.

lassoan, 2019-03-12:
Thanks for the details. In general, lock should be acquired for the minimum necessary time. The only reason we might want to keep the mutex locked is to reduce overhead of acquiring the lock many times.

I would try to avoid invoking an event while holding a lock, as the chain reaction that an event may start is quite unpredictable.

I've also noticed that we used vtkMutexLock. This used to be non-recursive on Windows (i.e., the application hangs if the thread that already acquired the lock tries to re-acquire it), which is very prone to hang. It vtkMutexLock is still non-recursive then we should use something like vtkIGSIORecursiveCriticalSection.