openigtlink/SlicerOpenIGTLink #78: POLYDATA is not received properly

tokjun, 2019-03-20:
I tried to transfer a model node from one Slicer to another one using OpenIGTLink. It was working in Slicer 4.8 but not on 4.10.

Here's what I tried:
- Opened two Slicer instances (I'll call them Slicer 1 and Slicer 2)
- On Slicer 1, loaded a head CT example, and generated a skull model.
- Opened OpenIGTLinkIF on Slicer 2, created a connector node, configured it as a server, and activated.
- Opened OpenIGTLinkIF on Slicer 1, created a connector node, configure it as a client, and connected it to Slicer 2.
- Under I/O Configuration in OpenIGTLinkIF on Slicer 1, unfolded "Scene"->"IGTLConnector1"->"OUT", chose the model node, clicked "+" button, and clicked "Send".

When I did this using Slicer 4.8, the same model showed up on Slicer 2. However, the model did not show up when I used Slicer 4.10.1 although a model node was created in the scene.

Sunderlandkyl, 2019-03-20:
The missing polydata is fixed in https://github.com/IGSIO/OpenIGTLinkIO/commit/08a348c6266094e29bc5bb8e591a0191eee3c407, however there still seems to be a significant delay when receiving polydata that have a large number of polys (at least in Debug).

lassoan, 2019-03-21:
Awesome, thank you Kyle!

Performance in debug mode does not mean much - certain operations takes magnitudes more time in debug mode. If Release or RelWithDebInfo performance is bad then we could do a quick profiling to see if there is any obvious problem that we can easily address.

tokjun, 2019-03-21:
Thanks Kyle for the quick fix!

Sunderlandkyl, 2019-03-21:
Checked on nightly build. Performance is not a problem in release.

closed 2019-03-21 by Sunderlandkyl