openigtlink/SlicerOpenIGTLink #79: Missing Python methods from vtkMRMLImageMetaListNode

ageorgou, 2019-04-02:
Some of the methods of `vtkMRMLImageMetaListNode` appear not to be bound in Python. For example:
```python
>>> n = slicer.vtkMRMLImageMetaListNode()
>>> n.GetImageMetaElement(0)
Traceback (most recent call last):
  File "<console>", line 1, in <module>
AttributeError: 'vtkSlicerOpenIGTLinkIFModuleMRMLPython.vtkMRMLImag' object has no attribute 'GetImageMetaElement'
```

The same happens with `AddImageMetaElement`. This happens in Slicer 4.8.1 and 4.10.1. Is this a bug or are they not intended to be used from Python?

Thanks in advance!

Sunderlandkyl, 2019-04-02:
The ImageMetaElement struct is not wrapped, so any functions that use it as an argument/return type are also not wrapped.

We could potentially overload the method(s) to provide an alternate signature that could be wrapped:
```
AddImageMetaElement(Name, DeviceName, Modality, PatientName, PatientID, TimeStamp, Size, ScalarType)
```

ageorgou, 2019-04-02:
Is wrapping the struct difficult with the way Slicer does the Python bindings? Asking out of curiosity!

From a selfish point of view, we would like to use the `GetImageMetaElement` method, for which I suppose you could have one getter for each field, e.g.`GetImageMetaElementName(int)` etc. But that seems excessive and not very clean.

If there is no other way, I could submit a PR doing that, although we would probably first try to work around it by calling the widget code.

ageorgou, 2019-04-02:
From a bit of investigation, I take it that the problem is that the VTK wrappers ignore anything that involves nested structs (among some other things) (see e.g. [this](http://vtk.1045678.n5.nabble.com/Nested-structs-and-python-wrapping-td5730120.html))?

If that is the case, making `ImageMetaElement` its own non-nested class seems like an easy workaround. But perhaps there are other factors I'm missing, or you would prefer to avoid that?

adamrankin, 2019-04-02:
After a quick look at the code, moving it to the top level should be fine.

PR #80 (2019-04-10): BUG: Wrap ImageMetaElement by moving to top-level
Fixes #79. Replaces the nested struct `vtkMRMLImageMetaListNode::ImageMetaElement` by a non-nested class `vtkMRMLImageMetaElement`, so that methods that use it are wrapped by vtk/Slicer.

Do let me know if there are any style inconsistencies or if I have missed something! I added the copyright from a more recent file (I think), although `vtkMRMLImageMetaListNode.h`, where the code originally was, had a different header.
closed 2019-04-10 by Sunderlandkyl