openigtlink/SlicerOpenIGTLink #81: Two-way sync via OpenIGTLink

lassoan, 2019-04-05:
OpenIGTLinkIF does a great job for propagating changes between scenes for multi-user virtual reality. However, it is currently not possible to synchronize a transform (by setting it both as input and output node).

Two-way synchronization has the risk of infinite update loops, but it should not be difficult to prevent it. For example, we could store modified time just after an incoming node is updated. Outgoing node would only be updated if its modified time is more recent than that modified time (modified since it is received).

Sunderlandkyl, 2019-04-05:
I'm wondering how we can prevent desync between instances of Slicer as well.

Ex.
1. A sends transform to B.
1. B sends transform to A.
1. B receives transform from A.
1. A receives transform from B.
1. A and B now exist in different states.

Perhaps it's not an issue, since it only occurs when two users are interacting with the same object at the same time. It would be resolved to a single state if one user stops interacting.

lassoan, 2019-04-05:
The chance that B can send a transform after "A sends transform to B" but before "B receives transform from A" is almost negligible (there is just a few-millisecond window).

If there was a continuous stream of transforms at tens or hundreds times per second then it may happen but then resynchronization happens at the next update (within a fraction of a second).

Sunderlandkyl, 2019-04-08:
Node synchronization between 1 server and N clients should be implemented in: https://github.com/openigtlink/SlicerOpenIGTLink/pull/82 and https://github.com/IGSIO/OpenIGTLinkIO/pull/75

Sunderlandkyl, 2019-04-10:
Integrated in 102c3f3.

closed 2019-04-10 by Sunderlandkyl