openigtlink/SlicerOpenIGTLink #86: Builds on Windows are not copying DLLs correctly

markasselin, 2019-05-01:
DLLs from <SlicerOpenIGTLink-Build-Dir>/bin/Release are not copied to the inner build, so the OpenIGTLink module is not loaded correctly when Slicer launches in a custom build.

closed 2019-05-01 by markasselin

adamrankin, 2019-05-01:
Wait, is this fixed? This is a very annoying issue!

markasselin, 2019-05-01:
Slicer also loads dlls from /bin/Release, so this is not actually an error. It is inconvenient during development though

adamrankin, 2019-05-01:
I'd say this is something worth fixing!

Sunderlandkyl, 2019-05-01:
I think that the real problem is the fact that multiple --launcher-additional-settings cannot be supplied to the same Slicer.

AFAIK, their current location is the standard for extensions with third party libraries.