openigtlink/SlicerOpenIGTLink #91: Undefined Behavior in Slicer 4.10.1

adnanmunawar, 2019-06-11:
For some reason the openIGTLink
for Slicer is not working as expected in Slicer 4.10.1. We are trying to
send a transform over from a client and then printing it at the host.
Even though the host is receiving a packet, it's empty when printed. The
same, however, works on Slicer 4.6.2 and 4.8.

Sunderlandkyl, 2019-06-11:
How are you printing it?

Can you attach a Slicer log file?

adnanmunawar, 2019-06-13:
Just to confirm, does the OpenIGTLink module in slicer 4.10.* use OpenIGTLink Message Version 3?

If so, I think the issue is at our end, as we might be using the OpenIGTLink library version 1 to unpack the message on an ARM v7 processor. I am able to see the transforms using ROS OpenIGTLink so clearly the plugin is working but should I not still be able to unpack the message using OpenIGTLink Library version 1 or 2?

Sunderlandkyl, 2019-06-14:
SlicerOpenIGTLink uses the extended v2 header format from OpenIGTLink 3.0.
Messages with the extended header cannot be decoded using OpenIGTLink 1.0 or 2.0.

tokjun, 2019-10-22:
Related to this thread, I also noticed that SlicerOpenIGTLink cannot receive transforms from a version 1 client. For example, if I push transforms from the TrackerClient example in OpenIGTLink library, SlicerOpenIGTLink module gives:

Warning: In /work/Stable/S-4102-E-b/SlicerOpenIGTLink/OpenIGTLinkIF/MRML/vtkMRMLIGTLConnectorNode.cxx, line 506
vtkMRMLIGTLConnectorNode (0xba752e0): Incoming TRANSFORM device has no device name!

Shouldn't the module be able to receive version 1 message?

Sunderlandkyl, 2019-10-22:
It should. TrackerServer sends messages to Slicer fine (to a client node), so I wonder if there is an issue with receiving messages on a server type connection.

leochan2009, 2019-10-22:
In the line of 
https://github.com/openigtlink/OpenIGTLink/blob/1d6cd5ce750e3b4edba340646247d1b297ea43fa/Examples/Tracker/TrackerClient.cxx#L76
transMsg gets reinitialized, so the device name is empty. you can add this line after the reinitialization.
transMsg->SetDeviceName("Tracker");
same as the TrackerServer example:
https://github.com/openigtlink/OpenIGTLink/blob/1d6cd5ce750e3b4edba340646247d1b297ea43fa/Examples/Tracker/TrackerServer.cxx#L72

tokjun, 2019-10-23:
Ah, I wasn't aware of the change in the library. Thanks for pointing this out!