openigtlink/SlicerOpenIGTLink #99: Problems processing TDATA messages

mmWasDF, 2019-10-27:
Has anyone else experienced that the TDATA message type can not be received with the OIGTLinkIF / Remote ? TRANSFORM messages are working as expected, but i cannot figure out the steps i need to do in order to make it possible to send TDATA messages to a 3D-Slicer client. I'm using the stable version 4.10.2 of 3D-Slicer. Tutorials tell me this feature should work. 
When using MITK's OpenIGTLink module, everything works as expected. Because of this I'm assuming the problem is not my application, but 3D-Slicer. I experienced that even 'example (simulator) server' applications are only working when sending TRANSFORM messages.

lassoan, 2019-10-27:
There was a major rework of SlicerOpenIGTLink in Slicer-4.10.2 and some less-frequently used features were not available for a while. Test if TDATA messages are received correctly if you use a recent Preview Release of Slicer.

mmWasDF, 2019-10-28:
Thanks for your advice. Unfortunately it does also not work with the newest Slicer-4.11 Preview Release of Slicer.

leochan2009, 2019-10-28:
we need to update the OpenIGTLINKIF module with the newly added device.

mmWasDF, 2019-10-31:
I just tested it with my OpenIGTLink-Client and can confirm it works with the newest Slicer Preview Release Version. Thank you

closed 2019-10-31 by mmWasDF