openigtlink/SlicerOpenIGTLink #101: Transform Packet Odd Behavior

alexanderdsmith, 2019-11-13:
Hi there,

I'm using Slicer 4.8.1, and I am attempting to develop a Javascript Implementation of the IGTL packing algorithm for NodeJS. I have gotten to the point where Slicer is reading in my transform, name and all, with no warnings, but the transform received is empty. I believe I have correctly implemented the header with CRC and message correctly, but Slicer does not appear to be reading my message. I tried receiving a simple transform from Slicer, and it looks like the message I received does not follow the format I expected.

Here is an example transform message I am sending (no rotation, and +10mm in X, Y, and Z):
`00015452414e53464f524d0000006530000000000000000000000000000000000000000000000000000000000000000000307842ab4c3f4df49b3f8000000000000000000000000000003f8000000000000000000000000000003f800000412000004120000041200000
`
106 bytes

And here is that exact same transform sent to Javascript from IGTLink in Slicer:
`00015452414e53464f524d0000006530000000000000000000000000000000000000000000000000000000000000000000307842efbfbd4c3f4defbfbdefbfbd3fefbfbd00000000000000000000000000003fefbfbd00000000000000000000000000003fefbfbd0000412000004120000041200000`
118 bytes

I noticed that in a few places in the received message from Slicer, a 0xefbfbd seemed to appear seemingly out of nowhere, either added in, or replacing 0x80, or 0x9b. My checksum was calculated using the CRC64-ECMA182 standard.

I'm a bit lost, and any help with this issue would be greatly appreciated.
Thank you!
-Alex

alexanderdsmith, 2019-11-13:
Note: I've noticed that 0xefbfbd replaces the following bytes in the Transform header and message received from Slicer:

0xab, 0xf4, 0x9b, 0x80.

Everything else in my sent message is identical. I've attempted sending the exact same message that Slicer sends me back to Slicer, but the transform I receive is again just the identity.

Sunderlandkyl, 2019-11-13:
4.8.1 is an old version, and does not use SlicerOpenIGTLink.

The SlicerOpenIGTLink extension is used in the latest stable and preview releases (4.10.1 and 4.11.0).
Can you try with a newer version?

alexanderdsmith, 2019-11-13:
Hi,

Just tried with the newer version, and I am running into a similar problem.

Now packets I receive from Slicer are v2 data (OpenIGTLink version 3.0?).

Am I still able to send data via the v1 protocol in the new Slicer version? Or for that matter, is there a  well defined example packet that exists that I can use to test my server and make sure the data is not corrupted between my server and Slicer?

Thanks for your response!

alexanderdsmith, 2019-11-13:
`000253544154555300000000000000000000000000000000000000000000000000000000000000000000000000000000003f6cf69307e59e5ee9000c000a0000000a00000000000000000000000000000000000000000000000000000000000000000000000001000500030000000564756d6d7964756d6d7900025452414e53464f524d0000006530000000000000000000000000000000000000000000000000000000000000000000a4986cd1ff6c18f383000c002200000046000000003f8000000000000000000000000000003f8000000000000000000000000000003f8000000000000000000000000000000004000c00030000000f000e000300000000000c000300000000000f0003000000024d454d4c4e6f64654e616d654c696e6561725472616e73666f726d53545245414d5f49445f46524f4d53545245414d5f49445f544f5472616e73666f726d5374617475734f4b
`

Here's the packet I receive from Slicer now in case that helps.

alexanderdsmith, 2019-11-13:
Update: I've determined that my problem is with the CRC generation. My algorithm is working correctly, but I may be performing the CRC on the wrong dataset. I have been generating a CRC on the body component of my packet (in hex), and this appears to not be the way to do this. Any ideas?

Sunderlandkyl, 2019-11-13:
The packet that you receive now from Slicer is a STATUS message.
The format of the header is here: https://github.com/openigtlink/OpenIGTLink/blob/master/Documents/Protocol/header.md

The device type: 535441545553000000000000, resolves to = STATUS\0\0\0\0\0\0, so this is not the message you are looking for.

adamrankin, 2019-11-13:
> The type: "535441545553000000000000", resolves to = STATUS\0\0\0\0\0\0, so this is not the message you are looking for.

[image]

alexanderdsmith, 2019-11-13:
Hmm.. I think that status message includes my data though - the 3f800000 corresponding to my 1s in the matrix were present... Still working on the CRC to see if I can get that resolved

Thanks!!