openigtlink/SlicerOpenIGTLink #110: backward compatibility with release-2.0

leochan2009, 2021-03-15:
I tested the SlicerOpenIGTLink with the  example programs(TrackingData/ImageDatabaseServer) in openigtlink release-2.0. I need to run these commands in slicer python interactor before active the connector:
>>> a = slicer.mrmlScene.GetNodeByID("vtkMRMLIGTLConnectorNode1")
>>> a.SetOutgoingMessageHeaderVersionMaximum(1)
During the initial handshake with TrackingData/ImageDatabase servers, the connector sends a dummy status message to check openigtlink version of the server. if the SetOutgoingMessageHeaderVersionMaximum is not set, the dummy message will be header version 2 by default and the parsing will fail at the server side. The failure of parsing the status message propagate to following messages resulting no communication at all.  
Should we have a UI component to set the maximum version of the Outgoing Message Header?

Thanks!

lassoan, 2021-03-15:
It would be better to avoid complicating the GUI with more options (and the additional development and maintenance work).

Why is it a problem for the server that it receives a v2 message if it only supports v1? Shouldn't the server just drop the v2 message?

leochan2009, 2021-03-15:
In any openigtlink applications, they use : 
socket->Skip(headerMsg->GetBodySizeToRead(), 0); to skip any message that don't need to process or incompatible. such as the example application: 
https://github.com/openigtlink/OpenIGTLink/blob/release-2.0/Examples/ImageDatabaseServer/ImageDatabaseServer.cxx#L124
To skip the message body related to the header,  the GetBodySizeToRead() call is used. However, extracting the body size from the header message depends on the header version, see the code snippet here:
https://github.com/openigtlink/OpenIGTLink/blob/release-2.0/Source/igtlMessageBase.cxx#L150
so the m_BodySizeToRead is always zero, resulting the failure in skipping the message body. 
As the status message from SlicerOpenIGTLink is of HEADER_VERSION 2 by default, the older openigtlink application won't be able to skip the message.

leochan2009, 2021-03-15:
The unpack process in the release 2.0 can be modified to **NOT** checking the header version, which is the case for later releases. 
However, it is hard to update the legacy programs....

hallaf, 2021-04-20:
We upgraded from Slicer 4.8 to 4.11 and had two issues. We wrote our own code on client (robot) side, and had to update.
First needed to ignore STATUS package and read again to get fiducial marker data, which is fine.
But the header length also changed...was 58. 
The protocol document says the header size is 72 bytes, but on Slicer 4.11 we found it to be 70 bytes.
...not a super user so may have misinterpreted something, but those are our findings.