openigtlink/SlicerOpenIGTLink #113: IGT Protocol Document

hallaf, 2021-04-20:
The document says the header size is 72 bytes, but on Slicer 4.11 we found it to be 70 bytes.

lassoan, 2021-04-20:
According to [OpenIGTLink specification](http://openigtlink.org/developers/spec), the header size is 58 bytes. The 58-byte header is used in all OpenIGTLink versions.

The extended header size is variable. Currently 10 bytes content is specified for it (extended header size, metadata size, message id), the rest of it is reserved for future use. Current implementations that I know of send 12 bytes, while illustration in the OpenIGTLink specification shows 14 bytes, but it can be really anything (you need to use the size specified in the message).

I'll send a pull request to remove the example of `72` from the documentation.

hallaf, 2021-04-20:
Thanks Andras! 
I guess we were used to a constant size. A more careful reading of the spec would have helped us....my bad.
I have to confess I did like the example...just maybe a bit more emphasis that the size is now variable.
Thanks again!

closed 2021-04-30 by hallaf