openigtlink/SlicerOpenIGTLink #122: Update modules to use markups instead of annotations

lassoan, 2022-09-14:
Annotations module, which provides `vtkMRMLAnnotationROI` and `vtkMRMLAnnotationRuler` nodes have been deprecated since April 2021 and will be removed in Slicer-5.4 stable version early next year.
Since some modules in this extension use annotation nodes, these modules need to be updated to use markups nodes before the end of this year.

[This section](https://www.slicer.org/wiki/Documentation/Nightly/Developers/Tutorials/MigrationGuide/Slicer#Slicer_5.3%3A_Removed_Annotation_module) of the migration guide should help with implementing the changes and any questions can be asked at the [Remove Annotations module topic in the Slicer forum](https://discourse.slicer.org/t/remove-annotations-module/25271).

Impacted modules:
- PlusRemote\Logic\vtkSlicerPlusRemoteLogic.cxx 
- PlusRemote\MRML\vtkMRMLPlusRemoteNode.cxx 
- PlusRemote\qSlicerPlusRemoteModuleWidget.cxx

Sunderlandkyl, 2022-09-14:
Markups ROI nodes are now accepted/used by default in PlusRemote.

I'll remove the references to Annotations once nodes loaded from scenes are converted into Markups nodes automatically.

lassoan, 2022-09-14:
Awesome, thank you!

closed 2022-09-16 by lassoan