openigtlink/SlicerOpenIGTLink #133: How to use python code to call the slicer plug-in: OpenIGTLink in the external compiler pycharm?

ningw-git, 2024-08-16:
I am trying to do secondary development based on slicer, and I have made a new software using Python. Now I have a problem: how to install the slicer plug-in: OpenIGTLink in my software? Or how to use Python code in the compiler pycharm to call this plug-in? Please help me, as a beginner, I really need help.

Sunderlandkyl, 2024-08-16:
To connect to Plus/OpenIGTLink within Slicer with Python, you need to create and activate a vtkMRMLIGTLConnectorNode.
```python
connectorNode = slicer.mrmlScene.AddNewNodeByClass("vtkMRMLIGTLConnectorNode")
connectorNode.SetServerHostname("localhost")
connectorNode.SetServerPort(18944)
connectorNode.Start()
```

ningw-git, 2024-08-18:
Thank you very much for your answer. According to the content you provided, I successfully connected to the plus server.
Now I want to call another plug-in Fiducial Registration Wizard: I want to realize the registration function. How can I realize this registration function in my python code? It's really troublesome for you.

ningw-git, 2024-08-18:
[image]
This is the plugin. I want to use its function to realize the registration function. Please help.

> To connect to Plus/OpenIGTLink within Slicer with Python, you need to create and activate a vtkMRMLIGTLConnectorNode.
>
> ```python
> connectorNode = [...]

Sunderlandkyl, 2024-08-19:
You'll need to add a [vtkMRMLFiducialRegistrationWizardNode](https://github.com/SlicerIGT/SlicerIGT/blob/master/FiducialRegistrationWizard/MRML/vtkMRMLFiducialRegistrationWizardNode.h), add the from/to markups nodes, and then call ```slicer.modules.fiducialregistrationwizard.logic().UpdateCalibration(wizardNode)```.

closed 2024-08-19 by Sunderlandkyl

Sunderlandkyl, 2024-08-19:
Not a complete example, but something like:
```
wizardNode.SetAndObserveFromFiducialListNodeId(fromMarkupsNode.GetID())
wizardNode.SetAndObserveToFiducialListNodeId(toMarkupsNode.GetID())
wizardNode.SetOutputTransformNodeId(outputTransformNode.GetID())
slicer.modules.fiducialregistrationwizard.logic().UpdateCalibration(wizardNode)
```

ningw-git, 2024-08-19:
Thank you very much for your answer.
Thank you very much!

ningw-git, 2024-08-20:
I'm here to bother you again. I've encountered a new problem. First, I installed the OpenIGT plug-in in slicer, and then created a new plug-in script myself. According to your code, I can call this plug-in. However, I now want to add this function to the software developed by our team, so how can I install the OpenIGT plug-in on both sides of my software? My software is a secondary development based on slicer. I'm really sorry to bother you.

> Not a complete example, but something like:
>
> ```
> wizardNode.SetAndObserveFromFiducialListNodeId(fromMarkupsNode.GetID())
> wizardNode.SetAndObserveToFiducialListNodeId(toMarkupsNode.GetID()) [...]

ningw-git, 2024-08-20:
[image]
this is our project construct,thanks again!

Sunderlandkyl, 2024-08-20:
You are trying to communicate between Slicer and your software?
You would likely need to add an OpenIGTLink implementation to communicate between them.
I'm not sure I can provide a recommendation without knowing more about your software.

ningw-git, 2024-08-21:
Thank you for your reply. I want to know whether I can package and install the slicer plug-in into my software. Since I did not develop the software, I do not know the specific logic. I only know that the software is developed based on slicer. How can I package and install the plug-in in slicer? Please help.

lassoan, 2024-08-21:
> I want to know whether I can package and install the slicer plug-in into my software.

Yes, you can.

>  I only know that the software is developed based on slicer.

You may ask the person who develops that software to ask questions from us.

> How can I package and install the plug-in in slicer

You can use SlicerCAT - https://github.com/KitwareMedical/SlicerCAT