openigtlink/SlicerOpenIGTLink #141: Slicer crashes while receiving tracking data from Brainlab

tokjun, 2025-07-25:
Slicer crashes while receiving tracking data from Brainlab. Although it is not consistent, the crash occurs when a second instrument is brought into the field. 

~~~~
Thread 1 "SlicerApp-real" received signal SIGSEGV, Segmentation fault.
0x00007806e9576f35 in vtkMRMLScene::GetNodesByClass (this=0x5cb52e863f40, className=0x7806773ce079 "vtkMRMLIGTLConnectorNode", 
    nodes=std::vector of length 1, capacity 1 = {...}) at /home/junichi/slicer/Slicer/Libs/MRML/Core/vtkMRMLScene.cxx:1865
(gdb) bt
#0  0x00007806e9576f35 in vtkMRMLScene::GetNodesByClass
    (this=0x5cb52e863f40, className=0x7806773ce079 "vtkMRMLIGTLConnectorNode", nodes=std::vector of length 1, capacity 1 = {...})
    at /home/junichi/slicer/Slicer/Libs/MRML/Core/vtkMRMLScene.cxx:1865
#1  0x00007806773c3a74 in vtkSlicerOpenIGTLinkIFLogic::CallConnectorTimerHander (this=0x5cb5333884e0)
    at /home/junichi/slicer/Modules-Dev/SlicerOpenIGTLink/OpenIGTLinkIF/Logic/vtkSlicerOpenIGTLinkIFLogic.cxx:324
#2  0x000078067743be32 in qSlicerOpenIGTLinkIFModule::importDataAndEvents (this=0x5cb530dfcfd0)
    at /home/junichi/slicer/Modules-Dev/SlicerOpenIGTLink/OpenIGTLinkIF/qSlicerOpenIGTLinkIFModule.cxx:218
#3  0x000078067743dc1c in qSlicerOpenIGTLinkIFModule::qt_static_metacall
    (_o=0x5cb530dfcfd0, _c=QMetaObject::InvokeMetaMethod, _id=3, _a=0x7fff65a4ef10)
    at /home/junichi/slicer/Modules-Dev/SlicerOpenIGTLink-build/inner-build/OpenIGTLinkIF/moc_qSlicerOpenIGTLinkIFModule.cpp:90
#4  0x00007806f5712dbf in ??? () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00007806f571710d in QTimer::timeout(QTimer::QPrivateSignal) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007806f570624b in QObject::event(QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#7  0x00007806f636bd45 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007806f78ef254 in qSlicerApplication::notify (this=0x7fff65a4f480, receiver=0x5cb530dff030, event=0x7fff65a4f130)
    at /home/junichi/slicer/Slicer/Base/QTGUI/qSlicerApplication.cxx:576
#9  0x00007806f56d8118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007806f57345ab in QTimerInfoList::activateTimers() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#11 0x00007806f5734f11 in ??? () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007806c93145c5 in ??? () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#13 0x00007806c9373737 in ??? () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#14 0x00007806c9313a63 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#15 0x00007806f5735279 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
    at /lib/x86_64-linux-gnu/libQt5Core.so.5
#16 0x00007806f56d6a7b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#17 0x00007806f56df3e8 in QCoreApplication::exec() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#18 0x00007806f6b0fdd0 in qSlicerCoreApplication::exec ()
    at /home/junichi/slicer/Slicer/Base/QTCore/qSlicerCoreApplication.cxx:1072
#19 0x00005cb516b83536 in (anonymous namespace)::SlicerAppMain (argc=1, argv=0x7fff65a4f628)
    at /home/junichi/slicer/Slicer/Applications/SlicerApp/Main.cxx:65
--Type <RET> for more, q to quit, c to continue without paging--
#20 0x00005cb516b83686 in main (argc=1, argv=0x7fff65a4f628)
    at /home/junichi/slicer/Slicer/Base/QTApp/qSlicerApplicationMainWrapper.cxx:56
~~~~

leochan2009, 2025-07-25:
i can reproduce the error. The crash always happen when mode2 starts:
[https://github.com/tokjun/IGTLTrackingDataTest/blob/brainlab-test/TrackingDataServer.cxx#L279](url)

leochan2009, 2025-07-27:
if a new tracking data is added, its name is in the key of the tBundleNode TrackingDataList,
 the tBundleNode element will be nElements + 1, but nElements is still not changed, resulting in the searching not completed.  the found flag will be false though the transformnode is there in the tBundleNode. resulting the Node->Delete() in UpdateTransformNode() function call. In the TrackingDataList of type std::map, the position of the element is determined by the key’s sorted order, not by when it is inserted. For example "Instruement"< "Pointer"<"Reference", Instruement will be at first position.

leochan2009, 2025-07-27:
Should be fixed by the latest commit. let me know if it works!

tokjun, 2025-07-29:
Thank you @leochan2009 ! I will test it hopefully this Friday with the system.